PiSupply/RAK811-Arduino #7: Downlink reference

nkebx2, 2020-09-10:
Hi, thank you for sharing this library. I've successfully sent data to TTN using JoinNetworkABP sketch. However, when I was trying to downlink message from TTN, my device does not show any message receiving from TTN. 

May I know if there's any example for TTN downlink application? Thanks!

ryanteck, 2020-09-10:
We don't currently have an example with downlink shown, however I'll see if I can try it out and what lines are added if it is easily done.

nkebx2, 2020-09-11:
Thanks for the reply, looking forward to your update

ryanteck, 2020-09-11:
Not had time today to look at this sorry as have had to resolve another issue.

It looks like the library has a function for it, so it'll be a case of testing it out.

https://github.com/PiSupply/RAK811-Arduino/blob/master/RAK811.cpp#L277

nkebx2, 2020-09-12:
Great. I will test it out, appreciate your help!