PiSupply/RAK811-Arduino #8: Changing band/region/frequency

spencerbh, 2020-09-10:
https://github.com/PiSupply/RAK811-Arduino/blob/4f3c83e4f2bb2d0e25beb9e2cbeb3ef82ffa45b3/RAK811.cpp#L55

This command is referenced in https://learn.pi-supply.com/make/getting-started-with-the-lora-node-arduino-shield/#setting-your-frequency as the method to change region/frequency/band. 

I get ERROR-1 when trying this. I am confused as to how this works as this command takes no arguments. I believe I am not able to get my node to connect to my gateway because of this issue.

ryanteck, 2020-09-10:
Hi @spencerbh ,

Just to confirm you're running the serial passthrough sketch so the Arduino acts as a passthrough. Then got ERROR-1 when sending a new frequency?

Which region are you in / trying to set it to?

spencerbh, 2020-09-10:
thanks for the response, New to Arduino, yes I am running the serial pass through sketch, im trying to set it to US915, but even if I try with EU868 I get the same ERROR-1

ryanteck, 2020-09-10:
Hmm, that is indeed odd. Sending "at+band=US915" should be working via the serial console.

Just to check, did you run the setbaudrate sketch first to begin with?

spencerbh, 2020-09-10:
for context when I attempt to disable the channels unused by TTN, the first line, `at+set_config=ch_mask:0,FF00`, returns OK, but the second line, `at+set_config=ch_mask:1,0000`, returns ERROR-1

spencerbh, 2020-09-10:
> Hmm, that is indeed odd. Sending "at+band=US915" should be working via the serial console.
> 
> Just to check, did you run the setbaudrate sketch first to begin with?

yes, I do that first and get lots of messages confirming `Send data ok!`

ryanteck, 2020-09-10:
Ok that does sound very odd, I'll get my node shortly and try it out myself.

nkebx2, 2020-09-11:
> 
> 
> > Hmm, that is indeed odd. Sending "at+band=US915" should be working via the serial console.
> > Just to check, did you run the setbaudrate sketch first to begin with?
> 
> yes, I do that first, then run the JoinNetworkABP and get lots of messages confirming `Send data ok!`

Have you tried changing from the default "Newline" to "Both NL & CR" before sending "at+band=US915" on the serial monitor? It works for my case.

spencerbh, 2020-09-11:
Big thanks that was the trick :+1:

ryanteck, 2020-09-11:
Oh that's interesting to know that it's not default! On mine it is (Or I've maybe changed it without realising.

I'll get the documentation updated to highlight that needs setting. Thanks @nkebx2

ryanteck, 2020-09-11:
Updated the tutorial, sorry about missing that out @spencerbh and thanks again @nkebx2 for pointing it out.

closed 2020-09-11 by ryanteck